orbbec/OrbbecSDK #152: Any plans to produce an ARM version for Windows?

dbruning, 2024-11-02:
We would like to run our app on the new Copilot+ PCs, which use Snapdragon processors. Our app doesn't run fast enough under emulation, so we would like to build an ARM native version, but we would need ARM native version of the Orbbec SDK.
Thankyou!

zhonghong322, 2024-11-06:
ARM version for Windows is not currently part of our plans, but I will pass this feedback on to our product manager later.

dbruning, 2024-11-09:
Thankyou. Let me explain a bit more about why it's important.
Our app uses 3 x Femto Mega cameras, and the processing done by the app means that we have had to recommend high-spec computers for our customers. This increases the total cost of the solution, and puts it out of reach for some markets.
We've done the work to port our app to run natively on ARM, and it's fast enough that the new inexpensive Copilot+ PCs could be used with our app. (Note that the x64 version of our app is NOT fast enough when run under emulation on ARM).
The only stopping point is that we can't actually use the Orbbec cameras because there is no ARM version of the SDK. Therefore, lack of Windows ARM support is directly hurting our sales and therefore sales of Orbbec cameras.