orbbec/OrbbecSDK #154: OrbbecSDK Android wrapper status

adi6409, 2024-11-08:
Hello, this is not your usual issue.

Nobody seems to be checking on the OrbbecSDK-Android-Wrapper repo issues and nobody is updating it.
Is it still under maintenance? Is an update planed? 

The version is very far behind and lacks support for the Gemini 330 series.

Thank you

zhonghong322, 2024-11-11:
I am currently pushing the Android wrapper. It can be done today.

zhonghong322, 2024-11-11:
@adi6409   The Android wrapper based on Orbbec SDK v1 has been released and supports the Gemini 330 series. (https://github.com/orbbec/OrbbecSDK-Android-Wrapper/releases/tag/v1.10.3) The Android wrapper based on Orbbec SDK v2 is currently being adapted and is planned to be released by the end of this month.

zhonghong322, 2024-11-12:
https://github.com/orbbec/OrbbecSDK-Android-Wrapper/releases/tag/v1.10.3  
The Android wrapper has been updated.

closed 2024-11-12 by zhonghong322

zylo117, 2025-01-03:
> @adi6409 The Android wrapper based on Orbbec SDK v1 has been released and supports the Gemini 330 series. (https://github.com/orbbec/OrbbecSDK-Android-Wrapper/releases/tag/v1.10.3) The Android [...]

@zhonghong322 can you share when will android wrapper based on sdk v2 be released?