orbbec/OrbbecSDK #155: Can't Connect via Network

danzeeeman, 2024-11-09:
**Describe the bug**  
A clear and concise description of what the bug is.

**Reproduction Steps**  
Steps to reproduce the behavior:

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Log Output**  
\\2024-11-09 12:37:08.784 INFO  [250118] [loggerInit@18] **********************************************
2024-11-09 12:37:08.784 INFO  [250118] [loggerInit@19]  OrbbecViewer launched! Welcome!! 
2024-11-09 12:37:08.784 INFO  [250118] [loggerInit@20]  	- Version: V1.9.3
2024-11-09 12:37:08.784 INFO  [250118] [loggerInit@21]  	- Author: 
2024-11-09 12:37:08.784 INFO  [250118] [loggerInit@22]  	- E-Mail: 
2024-11-09 12:37:08.784 INFO  [250118] [loggerInit@23]  	- Company: orbbec
2024-11-09 12:37:08.784 INFO  [250118] [loggerInit@24]  	- Website: http://www.orbbec.com.cn/
2024-11-09 12:37:08.784 INFO  [250118] [loggerInit@25] **********************************************
[11/09 12:37:09.323903][info][250118][Context.cpp:67] Context created with config: /home/dantheman/Downloads/OrbbecViewer_v1.9.3_202401121537_linux_x64_release/OrbbecSDKConfig_v1.0.xml
[11/09 12:37:09.323909][info][250118][Context.cpp:72] Context work_dir=/home/dantheman/Downloads/OrbbecViewer_v1.9.3_202401121537_linux_x64_release
[11/09 12:37:09.324342][info][250118][LinuxPal.cpp:112] Create PollingDeviceWatcher!
[11/09 12:37:09.324356][info][250118][DeviceManager.cpp:15] Current found device(s): (0)
[11/09 12:37:09.324373][info][250118][DeviceManager.cpp:310] Enable net device enumeration: true
[11/09 12:37:10.343064][info][250118][DeviceManager.cpp:15] Current device(s) list: (1)
[11/09 12:37:10.343096][info][250118][DeviceManager.cpp:20] 	- Name: Femto Mega, PID: 0x0669, SN/ID: CL2N24100F5, Connection: Ethernet, MAC:48:B0:2D:9B:45:DE, ip:192.168.1.10
[11/09 12:37:31.460917][warning][250118][ObException.hpp:40] VendorTCPClient write data failed! socket=27, err_code=11
[11/09 12:37:31.461154][warning][250118][VendorNetDataPort.cpp:16] Execute failure! A libobsensor_exception has occurred!
	 - where:16#sendData
	 - msg:VendorTCPClient write data failed! socket=27, err_code=11
	 - type:N11libobsensor12io_exceptionE
[11/09 12:37:31.461205][error][250118][HostProtocol.cpp:420] Set property value failed! propertyId:97, rc:{statusCode: 2, respErrorCode: 65535, msg: send control transfer failed!}
[11/09 12:37:31.461228][warning][250118][ObException.hpp:40] Request failed, statusCode: 2, msg: send control transfer failed!
[11/09 12:37:31.461312][warning][250118][FemtoMegaNetDevice.cpp:223] Execute failure! A libobsensor_exception has occurred!
	 - where:223#createCommand
	 - msg:Request failed, statusCode: 2, msg: send control transfer failed!
	 - type:N11libobsensor12io_exceptionE
2024-11-09 12:37:31.461 ERROR [250118] [OBSensorServer::connectToDevice@289] function:getDevice
args:index
message:Create vendor command failed! Device no response or bad connection.
type:0

[11/09 12:37:49.455538][info][250136][DeviceManager.cpp:188] Device changed! removed: 1, added: 0
[11/09 12:37:49.455555][info][250136][DeviceManager.cpp:15] Removed device(s) list: (1)
[11/09 12:37:49.455570][info][250136][DeviceManager.cpp:20] 	- Name: Femto Mega, PID: 0x0669, SN/ID: CL2N24100F5, Connection: Ethernet, MAC:48:B0:2D:9B:45:DE, ip:192.168.1.10
[11/09 12:37:49.455586][info][250136][DeviceManager.cpp:15] Current device(s) list: (0)
2024-11-09 12:37:49.455 DEBUG [250136] [OBSensorServer::OBSensorServer@56] device removed:Femto Mega-uid:48:B0:2D:9B:45:DE
[11/09 12:38:14.499231][info][250136][DeviceManager.cpp:188] Device changed! removed: 0, added: 1
[11/09 12:38:14.499253][info][250136][DeviceManager.cpp:15] Current device(s) list: (1)
[11/09 12:38:14.499267][info][250136][DeviceManager.cpp:20] 	- Name: Femto Mega, PID: 0x0669, SN/ID: CL2N24100F5, Connection: Ethernet, MAC:48:B0:2D:9B:45:DE, ip:192.168.1.10
2024-11-09 12:38:14.499 DEBUG [250136] [OBSensorServer::OBSensorServer@65] device added:Femto Mega-uid:48:B0:2D:9B:45:DE
[11/09 12:38:42.628872][warning][250118][ObException.hpp:40] VendorTCPClient write data failed! socket=27, err_code=11
[11/09 12:38:42.628985][warning][250118][VendorNetDataPort.cpp:16] Execute failure! A libobsensor_exception has occurred!
	 - where:16#sendData
	 - msg:VendorTCPClient write data failed! socket=27, err_code=11
	 - type:N11libobsensor12io_exceptionE
[11/09 12:38:42.629016][error][250118][HostProtocol.cpp:420] Set property value failed! propertyId:97, rc:{statusCode: 2, respErrorCode: 65535, msg: send control transfer failed!}
[11/09 12:38:42.629038][warning][250118][ObException.hpp:40] Request failed, statusCode: 2, msg: send control transfer failed!
[11/09 12:38:42.629095][warning][250118][FemtoMegaNetDevice.cpp:223] Execute failure! A libobsensor_exception has occurred!
	 - where:223#createCommand
	 - msg:Request failed, statusCode: 2, msg: send control transfer failed!
	 - type:N11libobsensor12io_exceptionE
2024-11-09 12:38:42.629 ERROR [250118] [OBSensorServer::connectToDevice@289] function:getDevice
args:index
message:Create vendor command failed! Device no response or bad connection.
type:0

**System Environment:**

- OS: Name and version (e.g., Windows 10, Ubuntu 20.04).
- Architecture: x86_64, armv7, etc.
- Visual Studio/GCC：Version A.B.C

**Tool/SDK Version:**

- SDK: Version A.B.C
- OrbbecViewer：: Version A.B.C

**Device Information:**

- Model: Device model name.
- Firmware Version: Firmware revision number.

**Additional context**  
Add any other context about the problem here.

danzeeeman, 2024-11-09:
happens on both windows and linux haven't tried Mac.

darshanhegde, 2025-02-28:
Any resolution for this ? I get the same error on Ubuntu 22.04 with OrbbecViewer ( version:  V1.10.8 ) and Orbbec Femto Mega (Firmware: 1.2.7 )