orbbec/OrbbecSDK #158: Using getIntrinsic(), getDistortion() function error, (Error: Can not find matched camera param!)

leejae0720, 2024-11-26:
I have a question regarding the use of the Gemini 2L model.

I need to retrieve the camera's intrinsic and distortion values, so I tried using the getIntrinsic and getDistortion functions provided in the StreamProfile.hpp file. However, when I called these functions, I encountered the error: 'Error: Cannot find matched camera param!'. All related logs are as follows.

```
[11/26 17:52:11.412316][info][46030][LinuxPal.cpp:32] createObPal: create LinuxPal!
[11/26 17:52:11.494756][info][46030][LinuxPal.cpp:166] Create PollingDeviceWatcher!
[11/26 17:52:11.494854][info][46030][DeviceManager.cpp:15] Current found device(s): (1)
[11/26 17:52:11.494875][info][46030][DeviceManager.cpp:24]      - Name: Orbbec Gemini 2 L, PID: 0x0673, SN/ID: AY8M63P002L, Connection: USB3.0
[11/26 17:52:11.494906][info][46030][Pipeline.cpp:15] Try to create pipeline with default device.
[11/26 17:52:11.601524][info][46030][AbstractDevice.cpp:125]    - Firmware version: 1.4.53
[11/26 17:52:11.606745][info][46030][Gemini2LDevice.cpp:184] Gemini2 L device created! PID: 0x0673, SN: AY8M63P002L
[11/26 17:52:11.606770][info][46030][DeviceManager.cpp:150] Device created successfully! Name: Orbbec Gemini 2 L, PID: 0x0673, SN/ID: AY8M63P002L
[11/26 17:52:11.606810][info][46030][Pipeline.cpp:47] Pipeline created with device: {name: Orbbec Gemini 2 L, sn: AY8M63P002L}, @0x5F717D2C4330
[11/26 17:52:11.608357][info][46030][Gemini2LDevice.cpp:772] Color sensor has been created!
[11/26 17:52:11.609728][info][46030][VideoSensor.cpp:384] Query stream profile! size= 80, SensorType=OB_SENSOR_COLOR
[11/26 17:52:11.609745][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 30}
[11/26 17:52:11.609754][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 400, fps: 30}
[11/26 17:52:11.609761][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 400, fps: 15}
[11/26 17:52:11.609768][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 400, fps: 10}
[11/26 17:52:11.609775][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 400, fps: 5}
[11/26 17:52:11.609782][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 800, height: 600, fps: 30}
[11/26 17:52:11.609789][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 800, height: 600, fps: 15}
[11/26 17:52:11.609802][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 800, height: 600, fps: 10}
[11/26 17:52:11.609812][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 800, height: 600, fps: 5}
[11/26 17:52:11.609819][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 360, fps: 30}
[11/26 17:52:11.609827][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 360, fps: 15}
[11/26 17:52:11.609836][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 360, fps: 10}
[11/26 17:52:11.609844][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 360, fps: 5}
[11/26 17:52:11.609853][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 30}
[11/26 17:52:11.609860][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 15}
[11/26 17:52:11.609868][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 10}
[11/26 17:52:11.609877][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 5}
[11/26 17:52:11.609890][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 800, fps: 30}
[11/26 17:52:11.609904][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 800, fps: 15}
[11/26 17:52:11.609915][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 800, fps: 10}
[11/26 17:52:11.609928][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 800, fps: 5}
[11/26 17:52:11.609941][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 400, fps: 30}
[11/26 17:52:11.609953][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 400, fps: 15}
[11/26 17:52:11.609962][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 400, fps: 10}
[11/26 17:52:11.609969][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 400, fps: 5}
[11/26 17:52:11.609977][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 800, height: 600, fps: 30}
[11/26 17:52:11.609986][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 800, height: 600, fps: 15}
[11/26 17:52:11.609993][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 800, height: 600, fps: 10}
[11/26 17:52:11.610005][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 800, height: 600, fps: 5}
[11/26 17:52:11.610018][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 360, fps: 30}
[11/26 17:52:11.610029][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 360, fps: 15}
[11/26 17:52:11.610040][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 360, fps: 10}
[11/26 17:52:11.610053][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 360, fps: 5}
[11/26 17:52:11.610069][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 15}
[11/26 17:52:11.610078][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 10}
[11/26 17:52:11.610089][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 5}
[11/26 17:52:11.610116][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 800, fps: 30}
[11/26 17:52:11.610123][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 800, fps: 15}
[11/26 17:52:11.610132][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 800, fps: 10}
[11/26 17:52:11.610141][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 800, fps: 5}
[11/26 17:52:11.610150][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_RGB888, width: 640, height: 400, fps: 30}
[11/26 17:52:11.610162][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_RGB888, width: 640, height: 400, fps: 15}
[11/26 17:52:11.610173][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_RGB888, width: 640, height: 400, fps: 10}
[11/26 17:52:11.610187][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_RGB888, width: 640, height: 400, fps: 5}
[11/26 17:52:11.610196][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_RGB888, width: 800, height: 600, fps: 30}
[11/26 17:52:11.610205][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_RGB888, width: 800, height: 600, fps: 15}
[11/26 17:52:11.610219][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_RGB888, width: 800, height: 600, fps: 10}
[11/26 17:52:11.610229][info][46030][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_RGB888, width: 800, height: 600, fps: 5}
[... cut ...]
        auto intrinsic = videoProfile->getIntrinsic();
        auto distortion = videoProfile->getDistortion();

        std::cout << streamName << " Stream - Width: " << width << ", Height: " << height << std::endl;
    } else {
        std::cout << "Stream profile for " << streamName << " is not a video profile!" << std::endl;
    }
}
```

For reference, if I remove the two sections below and run the code, it works correctly.
```
auto intrinsic = videoProfile->getIntrinsic();
auto distortion = videoProfile->getDistortion();
```

Best Regard

zhonghong322, 2024-12-02:
I will check the issues you have provided in your feedback

leejae0720, 2024-12-10:
Do I need to provide the code I wrote for you to review the issue? or has it already been confirmed that the variables are being appropriately output when using the corresponding function?