orbbec/OrbbecSDK #164: What file to run to open viewer?

shivanshpatel35, 2025-02-06:
The README suggests to run OBMultiStream as recommended [here](https://github.com/orbbec/OrbbecSDK/blob/401c2431e470590fdb34e783fed1e00aacc4b8d8/README.md?plain=1#L243). But when I built the folder in Ubuntu 22.04, I only see the following files in OrbbecSDK/build/bin

[image]

Am I looking at the wrong folder? Or does the README need to be updated?

zhonghong322, 2025-03-04:
Some samples depend on OpenCV, please install OpenCV first.

wzds2015, 2025-03-20:
> Some samples depend on OpenCV, please install OpenCV first.

Hi Zhonghong, I am having the same issue of Shivan. I installed the SDK following the guide. But I don't see OBMultiStream binary under bin/. Do you know why?

closed 2026-01-21 by zhonghong322