orbbec/OrbbecSDK #169: Update Orbbec Femto Mega via LAN

nheckmanHL, 2025-03-11:
I have my Orbbec cameras in a housing and i want to benefit from the newest feature but i dont want to disassembly the whole hosuing to update the Orbbec Firmware. Is there a possibility to update the Orbbec via LAN?

zhonghong322, 2025-03-28:
Femto Mega ，We will consider supporting network upgrade functionality in the next SDK version，I will update you once this feature is ready.

dbruning, 2025-06-26:
This would be amazing. Could we update from a Windows host, over the network, in this case? 
I note that [SDK v1.10.22](https://github.com/orbbec/OrbbecSDK/releases/tag/v1.10.22) says: "Fixed crash issue during network-based firmware update on Femto Mega/Femto Mega I." - does that mean it's supported now?

nheckmanHL, 2026-01-13:
Do you have any Updates about that topic? I can set the Orbbec into Recovery Mode via Software now. But i can‘t find the .bin Image.