orbbec/OrbbecSDK #171: 为什么识别不到Astra S，显示 Found a usb port without interface url! drop it!

Dingming3, 2025-03-24:
[03/24 21:25:35.746336][debug][25552][Context.cpp:30] Context creating, work_dir=D:\zzz\pyorbbecsdk-main\pyorbbecsdk-main\examples
[03/24 21:25:35.746400][debug][25552][Context.cpp:49] Config file version=1.1
[03/24 21:25:35.746416][debug][25552][FrameBufferManager.cpp:23] Max global frame buffer size updated! size=2048.000MB
[03/24 21:25:35.746423][info][25552][Context.cpp:68] Context created with config: D:\zzz\pyorbbecsdk-main\pyorbbecsdk-main\config\OrbbecSDKConfig_v1.0.xml
[03/24 21:25:35.746429][info][25552][Context.cpp:73] Work directory=D:\zzz\pyorbbecsdk-main\pyorbbecsdk-main\examples, SDK version=v1.10.16-20241021-5113dad11
[03/24 21:25:35.746435][debug][25552][DeviceManager.cpp:30] DeviceManager init ...
[03/24 21:25:35.746439][info][25552][MfPal.cpp:105] createObPal: create WinPal!
[03/24 21:25:35.746443][debug][25552][MfPal.cpp:110] WmfPal init ...
[03/24 21:25:35.760227][debug][25552][MfPal.cpp:117] WmfPal created!
[03/24 21:25:35.760241][debug][25552][DeviceManager.cpp:34] Enable USB Device Enumerator ...
[03/24 21:25:35.771725][debug][25552][EnumeratorLibusb.cpp:324] queryDevicesInfo done!
[03/24 21:25:35.771744][warning][25552][MfPal.cpp:265] Found a usb port without interface url! drop it!(device url=\\?\USB#VID_2BC5&PID_0402#5&30741CDD&0&9#{A5DCBF10-6530-11D2-901F-00C04FB951ED} interface index=0)
[03/24 21:25:35.771832][debug][25552][MfPal.cpp:219] Create WinEventDeviceWatcher!
[03/24 21:25:35.771884][debug][25552][UsbDeviceEnumerator.cpp:78] No matched usb device found!
[03/24 21:25:35.771902][info][25552][DeviceManager.cpp:15] Current found device(s): (0)
[03/24 21:25:35.771907][debug][25552][DeviceManager.cpp:52] DeviceManager construct done!
[03/24 21:25:35.771928][info][25552][Pipeline.cpp:15] Try to create pipeline with default device.
[03/24 21:25:35.771934][warning][25552][ObException.cpp:5] No device found, fail to create pipeline!
[03/24 21:25:35.774642][debug][25552][Context.cpp:81] Context destroying ...
[03/24 21:25:35.774653][debug][25552][DeviceManager.cpp:56] DeviceManager destroy ...
[03/24 21:25:35.774656][debug][25552][DeviceManager.cpp:64] DeviceManager Destructors done
[03/24 21:25:35.920204][debug][25552][MfPal.cpp:128] WmfPal destroyed!
[03/24 21:25:35.920544][info][25552][Context.cpp:84] Context destroyed

zhonghong322, 2025-03-28:
unsupport Astra S

Dingming3, 2025-03-30:
so how do i use pycharm to fetch picture from AstraS？A different version SDK ？

dong-132, 2025-09-06:
Excuse me, have you solved it? I have encountered the same problem.