orbbec/OrbbecSDK #176: Cannot get IR data from Gemini 336L

youngsooroh, 2025-05-06:
**Describe the bug**  
Not able to get the IR output from the sensor.

A clear and concise description of what the bug is.

**Reproduction Steps**  
Steps to reproduce the behavior:
Install Gemini 336L.
Checkout, build and Install pyorbbec SDK v1.3.4 that uses OrbbecSDK v1.10.16
Run python examples/infrared_viewer.py

**Expected behavior**  
A clear and concise description of what you expected to happen.

Get IR data

**Log Output**  
Include relevant log entries or error messages that are related to the problem.You can attach log files if they are too large.

(venv) aims@yr-nitro:~/pyorbbec/pyorbbecsdk-1.3.4$ python examples/infrared_viewer.py 
[05/06 13:18:01.156481][info][12970][Context.cpp:68] Context created with config: /home/aims/pyorbbec/pyorbbecsdk-1.3.4/config/OrbbecSDKConfig_v1.0.xml
[05/06 13:18:01.156489][info][12970][Context.cpp:73] Work directory=/home/aims/pyorbbec/pyorbbecsdk-1.3.4, SDK version=v1.10.16-20241021-5113dad
[05/06 13:18:01.156497][info][12970][LinuxPal.cpp:32] createObPal: create LinuxPal!
[05/06 13:18:01.433265][info][12970][LinuxPal.cpp:166] Create PollingDeviceWatcher!
[05/06 13:18:01.433313][info][12970][DeviceManager.cpp:15] Current found device(s): (1)
[05/06 13:18:01.433324][info][12970][DeviceManager.cpp:24] 	- Name: Orbbec Gemini 336L, PID: 0x0807, SN/ID: CP82841000AN, Connection: USB3.2
[05/06 13:18:01.433383][info][12970][Pipeline.cpp:15] Try to create pipeline with default device.
[05/06 13:18:01.436804][info][12970][G2RDevice.cpp:292] Json Content: {
    "ExtensionInfo": 
    {
        "IspFwVer":"20240523"
    }
    
}
,dataSize:80
[05/06 13:18:01.436874][info][12970][G2RDevice.cpp:305] key: IspFwVer,value:20240523
[05/06 13:18:01.437749][info][12970][G2RDevice.cpp:1347] switchDepthWorkMode done! same mode. currentDepthMode: {name: Default, checksum(hex): 653f0e89f348e76dbbab95d4712e205b, optionCode: NORMAL}, targetDepthMode:{name: Default, checksum(hex): 653f0e89f348e76dbbab95d4712e205b, optionCode: NORMAL}
[05/06 13:18:01.438068][info][12970][AbstractDevice.cpp:125] 	- Firmware version: 1.2.81
[05/06 13:18:01.548299][info][12970][G2RDevice.cpp:81] G2RDevice created! PID:2055, SN:CP82841000AN, depthMode: {name: Default, checksum(hex): 653f0e89f348e76dbbab95d4712e205b, optionCode: NORMAL}
[05/06 13:18:01.548374][info][12970][DeviceManager.cpp:150] Device created successfully! Name: Orbbec Gemini 336L, PID: 0x0807, SN/ID: CP82841000AN
[05/06 13:18:01.548455][info][12970][Pipeline.cpp:47] Pipeline created with device: {name: Orbbec Gemini 336L, sn: CP82841000AN}, @0x60D307258940
[05/06 13:18:01.548531][warning][12970][ObException.cpp:5] No required type sensor found! sensorType: OB_SENSOR_IR
No required type sensor found! sensorType: OB_SENSOR_IR
[05/06 13:18:01.548894][info][12970][Pipeline.cpp:75] Pipeline destroyed! @0x60D307258940
[05/06 13:18:01.548914][info][12970][G2RDevice.cpp:88] G2RDevice destroyed! PID:2055, SN:CP82841000AN, depthMode: {name: Default, checksum(hex): 653f0e89f348e76dbbab95d4712e205b, optionCode: NORMAL}
[05/06 13:18:01.557913][info][12970][ObLibuvcDevicePort.cpp:70] uvc_close done.
[05/06 13:18:01.557960][info][12970][ObLibuvcDevicePort.cpp:71] ~ObLibuvcDevicePort done
[05/06 13:18:01.558608][info][12970][Context.cpp:84] Context destroyed

**System Environment:**

- OS: Name and version (e.g., Windows 10, Ubuntu 20.04). Ubuntu 22.04
- Architecture: x86_64, armv7, etc.  x86_64
- Visual Studio/GCC：Version A.B.C
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0

**Tool/SDK Version:**

- SDK: Version 1.10.16
- OrbbecViewer：: Version A.B.C

**Device Information:**

2025-05-06 13:10:07,586 - DEBUG - DeviceInfo(name=Orbbec Gemini 336L
, pid=2055
, vid=11205, uid=2-1-2
, serial_number=CP82841000AN
, firmware_version=1.2.81
, connection_type=USB3.2
, hardware_version=0.1
, supported_min_sdk_version=1.0.1
, device_type=1)

- Model: Device model name.
- Firmware Version: Firmware revision number.

**Additional context**  

We were able to obtain IR data using the Gemini2 sensor, but not from the Gemini336L sensor.

youngsooroh, 2025-05-06:
We need to resolve this issue urgently to check the suitability of this sensor on our devices and to meet our deployment schedule.

youngsooroh, 2025-05-06:
After digging up a documentation, I can get the data using lef and right IR

closed 2025-05-06 by youngsooroh

zhonghong322, 2025-05-07:
- The reason is that the Gemini 330 series does not have OB_SENSOR_IR. You can use the sample at https://github.com/orbbec/pyorbbecsdk/blob/main/examples/double_infrared_viewer.py.

- We recommend using the [v2-main branch](https://github.com/orbbec/pyorbbecsdk/releases/tag/v2.0.10), which is a Python wrapper based on the open-source SDK，the sample has been optimized and the IR sample and double IR sample have been unified.

youngsooroh, 2025-05-07:
Hi Zhong,

Thank you for the prompt response and the helpful suggestion. We've successfully resolved the issue by using the v2-main branch as recommended.

Thanks again for your assistance!