orbbec/OrbbecSDK #183: In unity crash related to body tracking (?)

SewerynKezerk, 2025-05-30:
**Describe the bug**  
The game crash with log message related to body tracking - both in editor and runtime. It happens in various not connected circumstances (maybe detection of new user - but for sure not always?).

**Reproduction Steps**  
I don't have steps to reproduce it

**Expected behavior**  
App shouldn't crash

**Log Output**  
```Removing user 0, ID: 14, Body: 0, Time: 215.7068
Adding user 0, ID: 14, Body: 0, Pos: (0.18, 0.10, 3.32), Time: 216.5235
Removing user 0, ID: 14, Body: 0, Time: 218.9568
Adding user 0, ID: 14, Body: 1, Pos: (-0.01, 0.08, 3.58), Time: 219.2235
Removing user 0, ID: 14, Body: 0, Time: 224.9569
Adding user 0, ID: 14, Body: 0, Pos: (0.10, 0.03, 3.61), Time: 226.557
Crash!!!
SymInit: Symbol-SearchPath: '.;C:\Users\User\Desktop\18967859553514-photo-booth-default-windows-desktop-64-bit-39;C:/Users/User/Desktop/18967859553514-photo-booth-default-windows-desktop-64-bit-39/Default Windows desktop 64-bit_Data/Plugins\x86_64;C:/Users/User/Desktop/18967859553514-photo-booth-default-windows-desktop-64-bit-39/Default Windows desktop 64-bit_Data/Plugins;C:\Users\User\Desktop\18967859553514-photo-booth-default-windows-desktop-64-bit-39;C:\WINDOWS;C:\WINDOWS\system32;', symOptions: 534, UserName: 'User'
OS-Version: 10.0.0
C:\Users\User\Desktop\18967859553514-photo-booth-default-windows-desktop-64-bit-39\Default Windows desktop 64-bit.exe:Default Windows desktop 64-bit.exe (00007FF6B4F90000), size: 692224 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 6000.0.24.64053
C:\WINDOWS\SYSTEM32\ntdll.dll:ntdll.dll (00007FFEF8800000), size: 2514944 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\System32\KERNEL32.DLL:KERNEL32.DLL (00007FFEF6F10000), size: 823296 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\System32\KERNELBASE.dll:KERNELBASE.dll (00007FFEF59C0000), size: 3981312 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\SYSTEM32\apphelp.dll:apphelp.dll (00007FFEF2A00000), size: 647168 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\Users\User\Desktop\18967859553514-photo-booth-default-windows-desktop-64-bit-39\UnityPlayer.dll:UnityPlayer.dll (00007FFD55930000), size: 34373632 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 6000.0.24.64053
C:\WINDOWS\System32\CRYPT32.dll:CRYPT32.dll (00007FFEF6190000), size: 1536000 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3775
C:\WINDOWS\System32\ucrtbase.dll:ucrtbase.dll (00007FFEF63D0000), size: 1355776 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\System32\USER32.dll:USER32.dll (00007FFEF76A0000), size: 1875968 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\System32\win32u.dll:win32u.dll (00007FFEF5990000), size: 159744 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\System32\GDI32.dll:GDI32.dll (00007FFEF6E50000), size: 176128 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\SYSTEM32\bcrypt.dll:bcrypt.dll (00007FFEF5870000), size: 155648 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\System32\gdi32full.dll:gdi32full.dll (00007FFEF6050000), size: 1253376 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\System32\msvcp_win.dll:msvcp_win.dll (00007FFEF5FA0000), size: 667648 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\System32\ole32.dll:ole32.dll (00007FFEF7C20000), size: 1699840 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\System32\combase.dll:combase.dll (00007FFEF7880000), size: 3686400 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\SYSTEM32\VERSION.dll:VERSION.dll (00007FFEEF120000), size: 45056 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.1150
C:\WINDOWS\System32\RPCRT4.dll:RPCRT4.dll (00007FFEF7F70000), size: 1138688 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\System32\msvcrt.dll:msvcrt.dll (00007FFEF7DD0000), size: 692224 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 7.0.26100.4061
C:\WINDOWS\System32\SHLWAPI.dll:SHLWAPI.dll (00007FFEF6850000), size: 430080 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\System32\SETUPAPI.dll:SETUPAPI.dll (00007FFEF71D0000), size: 4743168 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\System32\ADVAPI32.dll:ADVAPI32.dll (00007FFEF6C90000), size: 729088 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3775
C:\WINDOWS\System32\sechost.dll:sechost.dll (00007FFEF6BE0000), size: 679936 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\System32\SHELL32.dll:SHELL32.dll (00007FFEF8090000), size: 7524352 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\System32\wintypes.dll:wintypes.dll (00007FFEF5E20000), size: 1523712 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\System32\OLEAUT32.dll:OLEAUT32.dll (00007FFEF7E80000), size: 917504 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\System32\IMM32.dll:IMM32.dll (00007FFEF7660000), size: 196608 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\SYSTEM32\WINMM.dll:WINMM.dll (00007FFEE71D0000), size: 221184 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3624
C:\WINDOWS\SYSTEM32\OPENGL32.dll:OPENGL32.dll (00007FFE73C70000), size: 1097728 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3624
C:\WINDOWS\System32\WS2_32.dll:WS2_32.dll (00007FFEF6FE0000), size: 475136 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\SYSTEM32\IPHLPAPI.DLL:IPHLPAPI.DLL (00007FFEF4220000), size: 208896 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\SYSTEM32\WINHTTP.dll:WINHTTP.dll (00007FFEEE1C0000), size: 1171456 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\SYSTEM32\HID.DLL:HID.DLL (00007FFEF3EC0000), size: 61440 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.1150
C:\WINDOWS\SYSTEM32\d3d11.dll:d3d11.dll (00007FFEF1940000), size: 2514944 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\SYSTEM32\dxgi.dll:dxgi.dll (00007FFEF2DB0000), size: 1257472 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\SYSTEM32\GLU32.dll:GLU32.dll (00007FFED6900000), size: 184320 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.1150
C:\WINDOWS\SYSTEM32\dwmapi.dll:dwmapi.dll (00007FFEF2F10000), size: 221184 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\SYSTEM32\dxcore.dll:dxcore.dll (00007FFEF2F90000), size: 290816 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3624
C:\WINDOWS\SYSTEM32\directxdatabasehelper.dll:directxdatabasehelper.dll (00007FFEF2D20000), size: 393216 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\SYSTEM32\kernel.appcore.dll:kernel.appcore.dll (00007FFEF4870000), size: 106496 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\SYSTEM32\cfgmgr32.dll:cfgmgr32.dll (00007FFEF55D0000), size: 356352 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\System32\bcryptPrimitives.dll:bcryptPrimitives.dll (00007FFEF6520000), size: 626688 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\system32\uxtheme.dll:uxtheme.dll (00007FFEF2B90000), size: 716800 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
C:\WINDOWS\System32\shcore.dll:shcore.dll (00007FFEF6670000), size: 978944 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\SYSTEM32\windows.storage.dll:windows.storage.dll (00007FFEF3660000), size: 8740864 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.4061
C:\WINDOWS\SYSTEM32\profapi.dll:profapi.dll (00007FFEF58A0000), size: 192512 (result: 0), SymType: '-deferred-', PDB: '', fileVersion: 10.0.26100.3912
[... cut ...]
I'm was trying to use two different Unity plugins LightBuzz [link](https://assetstore.unity.com/packages/tools/integration/body-tracking-for-orbbec-femto-bolt-mega-azure-kinect-157915) and Rf Studio [link](https://assetstore.unity.com/packages/tools/integration/azure-kinect-and-femto-bolt-examples-for-unity-149700). Both from time to time crash with the same log message (as pasted above). I also noticed that when crash happens in OrbbecSDK.log.txt there is a drop of color frame drop to around 7 fps - not sure if this is a symptom or cause of the crash. This not very often though - sometime app is running for few hours with users using it and sometimes it crash after few minutes.

SewerynKezerk, 2025-07-02:
Hi, do you need any further information on that?

ekdrnqkdrn, 2025-10-29:
Hi, has this issue been resolved? I also get intermittent crashes with this log.

NinthGames, 2025-11-08:
I’m experiencing the same issue — occasional crashes in Unity. Sometimes the app runs fine for a day or two, but other times it crashes every 30 to 60 minutes. The stack trace always ends at `k4abt_tracker_destroy`.