orbbec/OrbbecSDK #184: uvc_open already opened on MacOS 14.6 with Femto Bolt

ashay-rt, 2025-06-03:
I downloaded the OrbbecSDK Release v1.10.16 to try to get the femto bolt working with my mac laptop. 

I managed to build Sample-HelloOrbbec (main thing I had to do was give explicit permission in the system settings to load the dynamic libraries), but then when running `./OBHelloOrbbec` I get this issue:

```
ashay@ashays-Air build % ./OBHelloOrbbec                      
SDK version: 1.10.16
SDK stage version: main
[06/03 11:39:58.936091][info][13774447][Context.cpp:68] Context created with config: default config!
[06/03 11:39:58.936167][info][13774447][Context.cpp:73] Work directory=/Users/ashay/Documents/ashay3/src/OrbbecSDK/examples/cpp/Sample-HelloOrbbec/build, SDK version=v1.10.16-20241021-c0329e356
[06/03 11:39:58.936323][info][13774447][MacPal.cpp:31] createObPal: create MacPal!
[06/03 11:39:58.941553][info][13774447][MacPal.cpp:125] Create PollingDeviceWatcher!
[06/03 11:39:58.941567][info][13774447][DeviceManager.cpp:15] Current found device(s): (1)
[06/03 11:39:58.941570][info][13774447][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: CL8A842016F, Connection: USB3.1
[06/03 11:39:58.941819][info][13774447][FemtoBoltUvcDevice.cpp:23] FemtoBoltUvcDevice init ...
[06/03 11:39:58.941990][info][13774447][FemtoBoltUvcDevice.cpp:122] Create command start!
[06/03 11:39:58.942718][warning][13774447][ObLibuvcDevicePort.cpp:51] uvc_open  path=1-2.1-4.2 already opened
[06/03 11:39:58.942745][warning][13774447][ObLibuvcDevicePort.cpp:54] uvc_open  path=1-2.1-4.2 failed,return res:$-3
[06/03 11:39:58.944150][info][13774447][Context.cpp:84] Context destroyed
function:getDevice
args:index
message:uvc_open  path=1-2.1-4.2 failed,return res-3
type:0
```

This seems somewhat documented in other places:
- https://apple.stackexchange.com/questions/360851/add-access-to-the-macbook-camera-for-the-terminal-application
- https://github.com/orbbec/OrbbecSDK/issues/9

I can't kill the UVC assistant on mac. Has anyone gotten an orbbec camera to work on a MacOS that >= 12.0, which is when I think UVC stuff was introduced?

zhonghong322, 2025-06-21:
Femto Bolt is not supported on macOS because it depends on Microsoft’s Depth Engine library, for which there is no macOS version.

zhonghong322, 2025-06-21:
Femto Bolt is not supported on macOS

closed 2025-06-21 by zhonghong322